LibreELEC/LibreELEC.tv #9530: Disable connman online checks

sideeffect42, 2024-11-24:
### Describe the bug

On the LibreELEC 13.0 nightly builds I keep getting these errors in the journal:
```
kodi connmand[571]: Interface eth0 [ ethernet ] IPv4 online check to http://ipv4.connman.net/online/status.html failed: -22: Invalid argument
```

Everything works fine, though, I'm wondering if these online checks serve any purpose or if they could be disabled.

I think that sending unnecessary pings to connman developers can be problematic from a privacy perspective.

### Information
 - LibreELEC Version: 13.0-nightly
 - Hardware Platform: WeTek Play 2

chewitt, 2024-11-24:
It is possible to disable them with `cp /etc/connman/main.conf /storage/.config/connman_main.conf` then edit the file to turn them off and reboot.

I'm wondering if you have something like PiHole in the network blocking the request? .. because no issue here:

```
Nov 14 22:11:38 WP2 connmand[508]: Interface eth0 [ ethernet ] IPv4 online check to http://ipv4.connman.net/online/status.html succeeded
Nov 14 22:11:38 WP2 connmand[508]: Interface eth0 [ ethernet ] state is online
```

chewitt, 2024-11-24:
NB: Please direct further take further Q&A to the forum.

closed 2024-11-24 by chewitt

sideeffect42, 2024-11-24:
> It is possible to disable them with `cp /etc/connman/main.conf /storage/.config/connman_main.conf` then edit the file to turn them off and reboot.

Thank you for this hint. I just implemented this as a workaround.

However the reason for me opening this issue was that I wondered if the online check actually serves any purpose?
Even with it failing on my device everything else seems to function just fine.

Couldn't it be disabled by default as a more privacy friendly choice?

> I'm wondering if you have something like PiHole in the network blocking the request? .. because no issue here:

No, the request is not blocked.
I tried to `curl` the URL (on the play2) I get the "empty HTML page" back.