LibreELEC/LibreELEC.tv #9562: [BUG] RK3399   (eg: `NanoPC T4`)  has stopped working after Kernel 6.x

ihipop, 2024-12-10:
### Describe the bug
RK3399  (eg: `NanoPC T4`)  has stopped working after Kernel 6.x
No HDMI output and boot

### How to reproduce
Steps to reproduce the behavior:
1. Go to https://libreelec.tv/downloads/rockchip/
2. Download LibreELEC 12.0 / LibreELEC 11.0 which has Kernel 6.x inside, burn it to sd card and boot from it
3. NO HDMI output. No boot either

### Information
 - LibreELEC Version: 12.0
 - Hardware Platform: `NanoPC T4`
### Log file
No log because I can't ssh into it because ssh is disabled when a fresh install

### Context
LibreELEC 10.0 which has kernel 5.x works

heitbaum, 2024-12-10:
Please retest with LE13 nightlies - latest currently is 2024-12-09. https://test.libreelec.tv/13.0/Rockchip/nanopc-t4/ (you can boot with ssh as a kernel option to force start of the ssh.) Best move the support request to https://forum.libreelec.tv/ where many others will be able to assist.

closed 2024-12-10 by heitbaum

ihipop, 2024-12-10:
moved to https://forum.libreelec.tv/thread/29289-rk3399-eg-nanopc-t4-has-stopped-working-after-kernel-6-x/