LibreELEC/LibreELEC.tv #9606: [BUG] WeTek Play 2: 5.1ch AC3 no audio

sideeffect42, 2024-12-22:
### Describe the bug
When playing a video with a 5.1 channel AC3 audio stream, no audio can be heard on the HDMI output.
Enabling/disabling passthrough makes no difference.

Switching to a 2.0 audio stream (if the video has multiple audio streams) restores audio.

### How to reproduce
Steps to reproduce the behavior:
1. Play a video with a 5.1 channel AC3 audio stream
2. Hear no audio

### Information
 - LibreELEC Version: 13.0 (nightly-20241220-4d9ff19)
 - Hardware Platform: WeTek Play 2

### Log file
```
2024-12-22 10:58:05.352 T:1120     info <general>: CVideoSyncGbm::Setup: opened (fd:0 crtc:43 seq:8120 ns:249728691914:249737754569)
2024-12-22 10:58:05.362 T:877      info <general>: VideoPlayer::OpenFile: some.mkv
2024-12-22 10:58:05.364 T:1121     info <general>: Creating InputStream
2024-12-22 10:58:05.463 T:1121     info <general>: Creating Demuxer
2024-12-22 10:58:05.492 T:1121     info <general>: Opening stream: 0 source: 256
2024-12-22 10:58:05.492 T:1121     info <general>: [WHITELIST] Searching the whitelist for: width: 720, height: 576, fps: 50.000, 3D: false
2024-12-22 10:58:05.493 T:1121     info <general>: Display resolution ADJUST : 1920x1080 @ 50.000000 Hz (21) (weight: 0.000)
2024-12-22 10:58:05.767 T:877      info <general>: VideoPlayer: OnLostDisplay received
2024-12-22 10:58:05.767 T:877      info <general>: GLES: Maximum texture width: 4096
2024-12-22 10:58:05.768 T:1121     info <general>: Creating video codec with codec id: 2
2024-12-22 10:58:05.768 T:1121     info <general>: CDVDVideoCodecDRMPRIME::Open - using decoder V4L2 mem2mem MPEG2 decoder wrapper
2024-12-22 10:58:05.769 T:1121     info <general>: CDVDVideoCodecDRMPRIME::Open - unable to open codec
2024-12-22 10:58:05.769 T:1121     info <general>: CDVDVideoCodecDRMPRIME::Open - using decoder MPEG-2 video
2024-12-22 10:58:05.771 T:1121     info <general>: Creating video thread
2024-12-22 10:58:05.772 T:1127     info <general>: running thread: video_thread
2024-12-22 10:58:05.772 T:1121     info <general>: Opening stream: 2 source: 256
2024-12-22 10:58:05.772 T:1121     info <general>: Finding audio codec for: 86019
2024-12-22 10:58:05.773 T:1121     info <general>: CDVDAudioCodecFFmpeg::Open() Successful opened audio decoder ac3
2024-12-22 10:58:05.773 T:1121     info <general>: Creating audio thread
2024-12-22 10:58:05.773 T:1128     info <general>: running thread: CVideoPlayerAudio::Process()
2024-12-22 10:58:05.774 T:1121     info <general>: Opening stream: 6 source: 256
2024-12-22 10:58:06.277 T:877      info <general>: VideoPlayer: OnResetDisplay received
2024-12-22 10:58:06.277 T:886      info <general>: CActiveAESink::OpenSink - initialize sink
2024-12-22 10:58:06.277 T:886      info <general>: CAESinkALSA::Initialize - Attempting to open device "hdmi:CARD=WETEKPLAY2,DEV=0"
2024-12-22 10:58:06.278 T:1120     info <general>: CVideoSyncGbm::Setup: opened (fd:0 crtc:43 seq:8173 ns:250662003841:250663125736)
2024-12-22 10:58:06.282 T:886      info <general>: CAESinkALSA::Initialize - Opened device "hdmi:CARD=WETEKPLAY2,DEV=0,AES0=0x04,AES1=0x82,AES2=0x00,AES3=0x00"
2024-12-22 10:58:06.282 T:886      info <general>: CAESinkALSA::InitializeHW - Your hardware does not support AE_FMT_FLOAT, trying other formats
2024-12-22 10:58:06.282 T:886      info <general>: CAESinkALSA::InitializeHW - Using data format AE_FMT_S24NE4
2024-12-22 10:58:06.328 T:1128     info <general>: CVideoPlayerAudio: display reset occurred, checking for passthrough
2024-12-22 10:58:06.330 T:1128     info <general>: CDVDAudioCodecFFmpeg::Open() Successful opened audio decoder ac3
2024-12-22 10:58:06.330 T:1128     info <general>: Creating audio stream (codec id: 86019, channels: 6, sample rate: 48000, no pass-through)
2024-12-22 10:58:06.333 T:886      info <general>: CActiveAESink::OpenSink - initialize sink
2024-12-22 10:58:06.533 T:886      info <general>: CAESinkALSA::Initialize - Attempting to open device "hdmi:CARD=WETEKPLAY2,DEV=0"
2024-12-22 10:58:06.538 T:886      info <general>: CAESinkALSA::Initialize - Opened device "hdmi:CARD=WETEKPLAY2,DEV=0,AES0=0x04,AES1=0x82,AES2=0x00,AES3=0x02"
2024-12-22 10:58:06.539 T:886      info <general>: CAESinkALSA::InitializeHW - Your hardware does not support AE_FMT_FLOAT, trying other formats
2024-12-22 10:58:06.539 T:886      info <general>: CAESinkALSA::InitializeHW - Using data format AE_FMT_S24NE4
2024-12-22 10:58:06.540 T:886      info <general>: CAESinkALSA::Initialize - Attempting to open device "default"
2024-12-22 10:58:06.545 T:886      info <general>: CAESinkALSA::Initialize - Opened device "default"
2024-12-22 10:58:06.545 T:886      info <general>: CAESinkALSA::InitializeHW - Your hardware does not support AE_FMT_FLOAT, trying other formats
2024-12-22 10:58:06.546 T:886      info <general>: CAESinkALSA::InitializeHW - Using data format AE_FMT_S24NE4
2024-12-22 10:58:06.547 T:886     error <general>: CActiveAESink::OpenSink - no sink was returned
2024-12-22 10:58:06.547 T:885     error <general>: ActiveAE::InitSink - returned error
2024-12-22 10:58:06.547 T:885   warning <general>: CActiveAE::StateMachine - signal: 11 from port: OutputControlPort not handled for state: 2
2024-12-22 10:58:06.547 T:885   warning <general>: CActiveAE::StateMachine - signal: 12 from port: OutputControlPort not handled for state: 2
2024-12-22 10:58:07.103 T:877      info <general>: [WHITELIST] Searching the whitelist for: width: 720, height: 576, fps: 25.000, 3D: false
2024-12-22 10:58:07.104 T:877      info <general>: Display resolution ADJUST : 1920x1080 @ 50.000000 Hz (21) (weight: 0.000)
2024-12-22 10:58:08.896 T:1128    error <general>: CDVDAudio::AddPacketsRenderer - timeout adding data to renderer
2024-12-22 10:58:19.760 T:1128     info <general>: Skipped 5 duplicate messages..
2024-12-22 10:58:19.759 T:1128    error <general>: CDVDAudio::AddPacketsRenderer - timeout adding data to renderer
2024-12-22 10:58:30.622 T:1128     info <general>: Skipped 5 duplicate messages..
2024-12-22 10:58:30.622 T:1128    error <general>: CDVDAudio::AddPacketsRenderer - timeout adding data to renderer
2024-12-22 10:58:41.485 T:1128     info <general>: Skipped 5 duplicate messages..
2024-12-22 10:58:41.485 T:1128    error <general>: CDVDAudio::AddPacketsRenderer - timeout adding data to renderer
2024-12-22 10:58:47.357 T:877      info <general>: Skipped 3 duplicate messages..
```

chewitt, 2024-12-22:
Our support forum is over here: https://forum.libreelec.tv

closed 2024-12-22 by chewitt

sideeffect42, 2024-12-22:
Apologies, wasn't aware this is a support request...

sideeffect42, 2024-12-22:
In case someone else is having the same problem: no audio occurs whenever the output channels are set to anything higher than 2.0, HDMI or S/PDIF.

It is not related to AC3. Also the menu navigation sounds stop when "Stereo Upmix" is enabled and output channels are set to anything higher than 2.0.