LibreELEC/LibreELEC.tv #9618: [BUG] Aliases do not work in systemd service scripts

dtechsrv, 2024-12-27:
Aliases in the `[Install]` section of systemd service scripts do not work on LE12. _(I haven't tried another version yet.)_

For example, `systemctl stop docker` does not work for Docker. It doesn't show any error, but it has no effect.
However, `systemctl stop service.system.docker` works flawlessly.

The strange part is that if I start it with `systemctl start docker`, then `systemctl stop docker` works.
However, if I start it with the alias name and try to stop it with the full service name, it also does not work.

```
Dec 27 08:42:06 RPi4 systemctl[1648]: Created symlink /storage/.config/system.d/service.system.docker.service â†’ /storage/.kodi/addons/service.system.docker/system.d/service.system.docker.service.
Dec 27 08:42:06 RPi4 systemctl[1648]: Created symlink /storage/.config/system.d/docker.service â†’ /storage/.kodi/addons/service.system.docker/system.d/service.system.docker.service.
Dec 27 08:42:06 RPi4 systemctl[1648]: Created symlink /storage/.config/system.d/kodi.target.wants/service.system.docker.service â†’ /storage/.kodi/addons/service.system.docker/system.d/service.system.docker.service.
```

From what I've seen, the symlinks created at startup point to a good path, so there might be some deeper issue behind it.

heitbaum, 2024-12-27:
I can’t reproduce what you had described on `master`

```
nuc12:~ # systemctl status docker | cat
● service.system.docker.service - Docker Application Container Engine
     Loaded: loaded (/storage/.config/system.d/service.system.docker.service; enabled; preset: disabled)
     Active: active (running) since Wed 2024-12-25 09:11:03 AEDT; 2 days ago
 Invocation: 67100202033240a09e87b7131190cb5b
       Docs: https://docs.docker.com
    Process: 2103352 ExecStartPre=/storage/.kodi/addons/service.system.docker/bin/docker-config (code=exited, status=0/SUCCESS)
   Main PID: 2103355 (dockerd)
      Tasks: 792 (limit: 8192)
     Memory: 212.6M (peak: 229.6M)
        CPU: 12min 30.456s
     CGroup: /system.slice/service.system.docker.service
             ├─2103355 /storage/.kodi/addons/service.system.docker/bin/dockerd --exec-opt native.cgroupdriver=systemd --log-driver=journald --group=root --data-root=/storage/.kodi/userdata/addon_data/service.system.docker/docker --storage-driver=overlay2
             ├─2103370 containerd --config /var/run/docker/containerd/containerd.toml
             ├─2104338 /storage/.kodi/addons/service.system.docker/bin/containerd-shim-runc-v2 -namespace moby -id 3499218c1d284142a3af12b17dec1e2baffe7e99a577804a6a2b257296767bcc -address /var/run/docker/containerd/containerd.sock
             ├─2104345 /storage/.kodi/addons/service.system.docker/bin/containerd-shim-runc-v2 -namespace moby -id f7c3e81ddbcc5620ed5376e984e0dceef01375939de30c204a15163e105725ae -address /var/run/docker/containerd/containerd.sock
             └─2741371 /storage/.kodi/addons/service.system.docker/bin/containerd-shim-runc-v2 -namespace moby -id b350de0e30fe28d2c75f2bf956fbab6b88e5eba88b2129378f20c3dbeb0d7a72 -address /var/run/docker/containerd/containerd.sock

Dec 27 15:54:29 nuc12 dockerd[2103355]: time="2024-12-27T15:54:29.831386873+11:00" level=warning msg="failed to close stdin: NotFound: task 8eea1f12a18368d24b919b832373e5c71eaca619f6398e6f497b3ff6134dc049 not found: not found"
``` 

```
nuc12:~ # systemctl stop docker | cat
nuc12:~ # systemctl status docker | cat
○ service.system.docker.service - Docker Application Container Engine
     Loaded: loaded (/storage/.config/system.d/service.system.docker.service; enabled; preset: disabled)
     Active: inactive (dead) since Fri 2024-12-27 20:48:23 AEDT; 3s ago
   Duration: 2d 11h 36min 59.944s
 Invocation: 67100202033240a09e87b7131190cb5b
       Docs: https://docs.docker.com
    Process: 2103352 ExecStartPre=/storage/.kodi/addons/service.system.docker/bin/docker-config (code=exited, status=0/SUCCESS)
    Process: 2103355 ExecStart=/storage/.kodi/addons/service.system.docker/bin/dockerd --exec-opt native.cgroupdriver=systemd --log-driver=journald --group=root $DOCKER_DAEMON_OPTS $DOCKER_STORAGE_OPTS (code=exited, status=0/SUCCESS)
   Main PID: 2103355 (code=exited, status=0/SUCCESS)
   Mem peak: 229.6M
        CPU: 12min 31.469s

Dec 27 20:48:16 nuc12 dockerd[2103355]: time="2024-12-27T20:48:16.292073357+11:00" level=warning msg="Container failed to exit within 10s of kill - trying direct SIGKILL" container=f7c3e81ddbcc5620ed5376e984e0dceef01375939de30c204a15163e105725ae error="context deadline exceeded"
``` 

```
nuc12:~ # systemctl start service.system.docker.service | cat
nuc12:~ # systemctl status docker | cat
● service.system.docker.service - Docker Application Container Engine
     Loaded: loaded (/storage/.config/system.d/service.system.docker.service; enabled; preset: disabled)
     Active: active (running) since Fri 2024-12-27 20:49:39 AEDT; 2s ago
 Invocation: bd2fb3c4532540ca861d00e59095e118
       Docs: https://docs.docker.com
    Process: 3976485 ExecStartPre=/storage/.kodi/addons/service.system.docker/bin/docker-config (code=exited, status=0/SUCCESS)
   Main PID: 3976487 (dockerd)
      Tasks: 767 (limit: 8192)
     Memory: 183M (peak: 207.6M)
        CPU: 2.748s
     CGroup: /system.slice/service.system.docker.service
             ├─3976487 /storage/.kodi/addons/service.system.docker/bin/dockerd --exec-opt native.cgroupdriver=systemd --log-driver=journald --group=root --data-root=/storage/.kodi/userdata/addon_data/service.system.docker/docker --storage-driver=overlay2
             ├─3976503 containerd --config /var/run/docker/containerd/containerd.toml
             ├─3977451 /storage/.kodi/addons/service.system.docker/bin/containerd-shim-runc-v2 -namespace moby -id 3499218c1d284142a3af12b17dec1e2baffe7e99a577804a6a2b257296767bcc -address /var/run/docker/containerd/containerd.sock
             └─3977455 /storage/.kodi/addons/service.system.docker/bin/containerd-shim-runc-v2 -namespace moby -id f7c3e81ddbcc5620ed5376e984e0dceef01375939de30c204a15163e105725ae -address /var/run/docker/containerd/containerd.sock

Dec 27 20:49:39 nuc12 f7c3e81ddbcc[3976487]: User GID:    911
```

dtechsrv, 2024-12-27:
That's good news, but this bug exists in v12.0.1.
```
RPi4:~ # ps aux | grep docker
 2236 root      0:01 /storage/.kodi/addons/service.system.docker/bin/dockerd --exec-opt native.cgroupdriver=systemd --log-driver=journald --group=root --data-root=/storage/.kodi/userdata/addon_data/service.system.docker/docker --storage-driver=overlay2
 2244 root      0:07 containerd --config /var/run/docker/containerd/containerd.toml
 2653 root      0:00 grep docker
RPi4:~ # systemctl status docker | cat
○ docker.service - Docker Application Container Engine
     Loaded: loaded (/storage/.config/system.d/docker.service; alias)
     Active: inactive (dead)
       Docs: https://docs.docker.com
RPi4:~ # systemctl status service.system.docker | cat
● service.system.docker.service - Docker Application Container Engine
     Loaded: loaded (/storage/.config/system.d/service.system.docker.service; enabled; preset: disabled)
     Active: active (running) since Fri 2024-12-27 08:54:29 CET; 2h 15min ago
       Docs: https://docs.docker.com
    Process: 2233 ExecStartPre=/storage/.kodi/addons/service.system.docker/bin/docker-config (code=exited, status=0/SUCCESS)
   Main PID: 2236 (dockerd)
      Tasks: 20 (limit: 8192)
        CPU: 9.787s
     CGroup: /system.slice/service.system.docker.service
             ├─2236 /storage/.kodi/addons/service.system.docker/bin/dockerd --exec-opt native.cgroupdriver=systemd --log-driver=journald --group=root --data-root=/storage/.kodi/userdata/addon_data/service.system.docker/docker --storage-driver=overlay2
             └─2244 containerd --config /var/run/docker/containerd/containerd.toml

Dec 27 08:54:28 RPi4 dockerd[2236]: time="2024-12-27T08:54:28.404961571+01:00" level=info msg="[graphdriver] trying configured driver: overlay2"
Dec 27 08:54:28 RPi4 dockerd[2236]: time="2024-12-27T08:54:28.573621377+01:00" level=info msg="Loading containers: start."
Dec 27 08:54:28 RPi4 dockerd[2236]: time="2024-12-27T08:54:28.881907873+01:00" level=info msg="Default bridge (docker0) is assigned with an IP address 172.17.0.0/16. Daemon option --bip can be used to set a preferred IP address"
Dec 27 08:54:29 RPi4 dockerd[2236]: time="2024-12-27T08:54:29.018939183+01:00" level=info msg="Loading containers: done."
Dec 27 08:54:29 RPi4 dockerd[2236]: time="2024-12-27T08:54:29.107229417+01:00" level=warning msg="WARNING: No memory limit support"
Dec 27 08:54:29 RPi4 dockerd[2236]: time="2024-12-27T08:54:29.107302732+01:00" level=warning msg="WARNING: No swap limit support"
Dec 27 08:54:29 RPi4 dockerd[2236]: time="2024-12-27T08:54:29.107360713+01:00" level=info msg="Docker daemon" commit=f9522e5e96c3ab5a6b8a643d15a92700ca864da6 containerd-snapshotter=false storage-driver=overlay2 version=27.1.2
Dec 27 08:54:29 RPi4 dockerd[2236]: time="2024-12-27T08:54:29.107529898+01:00" level=info msg="Daemon has completed initialization"
Dec 27 08:54:29 RPi4 dockerd[2236]: time="2024-12-27T08:54:29.169663359+01:00" level=info msg="API listen on /var/run/docker.sock"
Dec 27 08:54:29 RPi4 systemd[1]: Started service.system.docker.service.
RPi4:~ # cat /etc/os-release
NAME="LibreELEC"
VERSION="12.0.1"
ID="libreelec"
VERSION_ID="12.0"
PRETTY_NAME="LibreELEC (official): 12.0.1"
HOME_URL="https://libreelec.tv"
BUG_REPORT_URL="https://github.com/LibreELEC/LibreELEC.tv"
BUILD_ID="7e0b15d49ae7beb086bd2fd00fbf5d107aa9f251"
LIBREELEC_ARCH="RPi4.aarch64"
LIBREELEC_BUILD="official"
LIBREELEC_PROJECT="RPi"
LIBREELEC_DEVICE="RPi4"
RPi4:~ #

```

heitbaum, 2024-12-27:
12 is running [255.4](https://github.com/LibreELEC/LibreELEC.tv/commit/4fecb340e5c650e6d13a2cce9a8a0296a1d7ba3a) current is https://github.com/systemd/systemd-stable/releases/tag/v255.16

whereas 13 is running 257.1.

Does updating to 255.16 fix your issue in LE12? If yes, and if you can identify the require patch, then it could be included in an LE12, probably not keen on updating to 255.16 in LE12 stable.

mglae, 2024-12-27:
Can confirm both of you: I've noticed too systemd aliases stopped working some time ago (in LE11 according to my quick test today) but do work again in LE13 (good news for me).

heitbaum, 2024-12-27:
> Can confirm both of you: I've noticed too systemd aliases stopped working some time ago (in LE11 according to my quick test today) but do work again in LE13 (good news for me).

LE11 was running 252.6. So issue looks to have been across quiet a few systemd releases

dtechsrv, 2024-12-27:
It seems this was the fix for this issue: https://github.com/systemd/systemd/pull/31816

dtechsrv, 2024-12-27:
I started an LE12 build for the RPi4 adding this patch. If it is successful, I will submit it in a PR.

dtechsrv, 2024-12-27:
The patch above really works, but a dependent patch was also needed, and the affected service had to be disabled and re-enabled afterwards. _(For a fresh installation, the latter is obviously not necessary.)_

closed 2024-12-28 by mglae