LibreELEC/LibreELEC.tv #9638: [BUG] No HDMI 2.1 (2160p) on Arrow Lake

ValeZAA, 2025-01-03:
### Describe the bug

Arrow Lake Core Uktra 7 has HDMI 2.1 48 gbit/s, all of it works just fine with LG C9 on windows. All 4k RGB 12 bit and even bitsreaming with VRR. On Linux neither debian nor LibreELEC has 4K120 option, not bitsreaming/pcm streaming works which is apparently a known bug.

### How to reproduce
Steps to reproduce the behavior:
1. Go to Setting Display
2. Observe there is no 3840x2160p option with 120.00 // 120/1.001 fps. BTW, would be nice to change it to display 60/1.001 instead of 59.94 and 30/1.001 instead of 29.97 in the case of alternate clock and not dedicated timing that just have low precision. Thanks.
3. See error

### Information
 - LibreELEC Version: [e.g. 9.2.1] latest nightly fron 31 December.
 - Hardware Platform: [e.g. RPi3] Intel Arrow Lake + Nvidia Turing

### Log file
 Not needed.

### Context

There may be no way to activate Arrow Lake HDMI 2.1 with open source drivers yet. Will investigate and maybe ask Intel.

antonlacon, 2025-01-03:
As far as I'm aware, none of the open source graphics drivers support HDMI 2.1. My understanding is your choices are a nvidia card with their closed drivers (which I don't remember their support for wayland or GBM which will effect future LE releases), trying your luck with a displayport -> hdmi 2.1 adapter that supports it (if available), or hoping for something to change in the future (but it's been years, so don't hold your breath). Some Intel cards appear to have a displayport -> hdmi 2.1 chip as part of the card, but if you've landed here, I don't believe that you do.

You can read more here: https://gitlab.freedesktop.org/drm/amd/-/issues/1417 It's for amd, but it's the same root problem: HDMI Forum doesn't want it to happen.

Closing as there isn't something that can be done on our end.

ValeZAA, 2025-01-03:
>but it's been years, so don't hold your breath

It has been 3 months since release of Arrow lake and 1 year since release of Meteor Lake, Intel did not support HDMI 2.1 before that.

chewitt, 2025-01-03:
If we are missing something it is also missing from the upstream Linux kernel source that we blindly package and the correct people to report issues to are the Intel developers that support the Intel GPU drivers.

LE master branch currently uses Linux 6.12.y so if the GPU is only recently released I would experiment with an image that uses the `drm-tip` repo: https://gitlab.com/freedesktop-mirror/drm-tip/ as this is based on Linux 6.13.y (currently rc5) and includes the latest changes from Intel developers. If that doesn't change anything report the issue to Intel developers telling them that you already tested `drm-tip` (else the first thing they will ask you to do is test it). Note that Intel devs generally have no clue about LibreELEC so it may also help to build/test that kernel on Debian which is more something their brains will understand.