LibreELEC/LibreELEC.tv #9642: [BUG] Aerial screensaver fails to activate during movie playback

khodess, 2025-01-04:
### Describe the bug
The Aerial screensaver activates when not playing back a movie but fails to activate when playing back a movie.  The black screensaver does not have this problem (although it fails to turn off the TV via CEC if activated during movie playback)

### How to reproduce
Steps to reproduce the behavior:
1. Set screensaver to Aerial
2. Set screensaver to activate after a minute
3. Pause a movie
4. Wait a minute for nothing to happen
5. Do the same with the black screensaver and see it activate

### Information
 - LibreELEC Version: 12.0.1
 - Hardware Platform: RPi4

### Log file
https://paste.libreelec.tv/trusting-yak.log

### Context
At 13h58, black screensaver, paused movie, a minute later black screensaver came on and did not power off TV

Immediately after at 13h59, in a Kodi menu, a minute later, black screensaver powered off TV, button presses on remote woke TV up

Changed to Aerial

At 14h01, aerial, paused movie, a minute later, nothing happened - no screensaver

Waited till 14h03, aerial, in a Kodi menu, a minute later at 14h04, screensaver turned off TV, button presses on remote did not wake TV up

prahal, 2025-01-11:
Maybe retitle to "[BUG] Aerial screensaver fails to activate while pausing movie playback" even though it is obvious the screensaver is not supposed to activate during playback itself.

heitbaum, 2025-02-27:
Please test using latest LE13 nightly image (use hdmi#0) hdmi#1 does not support Cec.
Please test with default black screensaver.
Potential duplicate of other cec and aerial issues. 
Can be reopened as one issue, with the scenarios if reproducible.

For normal cec support please use the forum.

closed 2025-02-27 by heitbaum

prahal, 2025-03-06:
I believe the comment about cec was a side note and not related to this bug report. SO not at all a duplicate of the other CEC bug report (except for the side note).
If the display is put to sleep by cec then the screensaver cannot be seen.
As the reporter told he already tried with the default black screensaver (which was not exhibiting the bug as the Aerial one did).
The scenarios are already detailed (ie the steps by steps instructions to reproduce) in the original report.

To reporter, please try to reproduce with LE13. Even though I believe a bug on LE12 is still a valid bug.

On my side, I reproduced that areal does indeed not show the pictures on latest LE12 for odroid c2.
Still, I noted that with Aerial and a paused video in the background, after 1 minutes a dim was applied and after 5 minutes a black screen. Could it be Aerial somewhat cannot display the pictures because there is already a paused video stream in the background? Ie it seems to me Aerial is doing a Keith Burns effect on the pictures and could be is using the Kodi video stream to do that.

heitbaum, 2025-03-06:
I can see screensaver.atv4 in the logs with nothing untoward. This addon is supported at https://kodi.wiki/view/Add-on:Aerial . 

If it is a LE bug (Please confirm in LE13 nightlies) - then best to raise a new Issue with the simplified issue. But on the look of it it will be an Addon or Kodi issue. Please use the "Addon Support forum" and/or the "LE support forum" before raising the bug here - as if it is code related - then it will need to be fixed at either https://kodi.wiki/view/Add-on:Aerial or Kodi.