LibreELEC/LibreELEC.tv #9645: [BUG] CEC does not turn off display when "System->Power Saving -> Put display to sleep when idle" is activated

khodess, 2025-01-04:
### Describe the bug
The display doesn't turn off when Put display to sleep when idle is activated.  LE just stops sending a display signal and the display has a no signal message appear

### How to reproduce
Steps to reproduce the behavior:
1. "System->Power Saving -> Put display to sleep when idle" is set to 5 minutes
2. Wait 5 minutes
3. Display not put to sleep, LE just stops sending any signal to display

### Information
 - LibreELEC Version: 12.0.1
 - Hardware Platform: RPi4

### Log file
https://paste.libreelec.tv/proud-sailfish.log

### Context
At 14h44, idled for 5 minutes
At 14h49, signal stopped being sent to display and display showed no signal on screen instead of turning off
At 14h50 Proved LE can turn off display by setting CEC settings to do so on screensaver activation and using the black screensaver

jemiller00, 2025-01-28:
I have been experimenting with the same setting as you and observed the same thing. I think it must just turn off the display adapter in the Pi. I had a Kill-A-Watt attached to the Pi to measure the used wattage and did notice that it went down about .2 to .4 of a watt. That's what makes me think it turned the display adapter off. I was hoping, as you, that it would turn off the TV as well. Presumably, using CEC to do it. I wasn't sure if that was a CEC function though. So, I figured that this was just the expected behavior. I'm using a Sony TV and my TV turns off after an additional timeout of not receiving a signal. It still basically does the job. Just not quite as nicely as what I wanted. It would be nice if LibreELEC was able to actually turn the TV off.

heitbaum, 2025-02-27:
Please test using latest LE13 nightly image (use hdmi#0) hdmi#1 does not support Cec.
Please test with default black screensaver.
Potential duplicate of other cec and aerial issues. 
Can be reopened as one issue, with the scenarios if reproducible.

For normal cec support please use the forum.

If it is a confirmed bug with LE - then a bug report can be logged.

closed 2025-02-27 by heitbaum