LibreELEC/LibreELEC.tv #9725: [BUG][LE13] GUI updater is nailed at 20250122

eldersixpack13, 2025-01-27:
### Describe the bug
- graph. updater does not see/contain current updates (should be 20250125-4cffb60, but is since some days 20250122-cbe5bdc)
- changelog is also uncurrent (nailed at 2025-01-22 (7b7c738): [#9713] )

### How to reproduce
Steps to reproduce the behavior:
1. try to update via GUI

### Information
 - LibreELEC Version: LE13 last nightly 
 - Hardware Platform: NUC13

### Screenshots

[image]
[image]

CvH, 2025-01-27:
the job was deactived by github, should work again

eldersixpack13, 2025-01-27:
yup, fixed.
Thanks

closed 2025-01-27 by eldersixpack13