LibreELEC/LibreELEC.tv #9726: HDMI CEC stopped working on Raspberry Pi 4B when hooked up to the second HDMI port after 12.0.2 update was installed

jemiller00, 2025-01-28:
I'm running LibreELEC on 3 Raspberry Pi 4Bs. I've been running version 12.0.1 and have been using them with Sony TVs. I use the Sony remotes that came with the TVs and I've been able to control LibreELEC using HDMI CEC. On one of the TVs, HDMI CEC stopped working. Around the same time I saw a popup on the screen saying that an update to LibreELEC was available. The HDMI cable to the TV was on what I'm assuming is the second HDMI port. The one that's to the right, furthest from the USB-C port used for power. I did a web search and found a page where someone else was having this problem a couple years ago and someone said that HDMI CEC doesn't work on the second HDMI port. It's been working fine on mine. I was worried that maybe it was a hardware problem. However, when LibreELEC updated from 12.0.1 to 12.0.2 on one of the other Raspberry Pis, the same problem happened on that one. If I hook the TV up using the HDMI port nearest the power cable on the Raspberry Pi, it seems to work fine. So, I'm thinking it's a bug in the new version of LibreELEC. I haven't updated the 3rd Pi that I have yet, but, I'm guessing that one will have the same problem.

Steps to reproduce the behavior:
1. Attach a micro-HDMI cable to the HDMI port furthest from the power connector on a Raspberry Pi 4B and attach it to a TV that support HDMI CEC, such as a Sony TV.
2. Test this on LibreELEC 12.0.1. It works.
3. Upgrade to LibreELEC 12.0.2. It doesn't work. The remote buttons don't control LibreELEC.
4. Hook the HDMI cable to the other HDMI port, the one nearest the power cable.
5. SSH into the Pi and reboot it.
6. Attempt to control LibreELEC using the TV remote. This time it works.

### Information
 - LibreELEC Version: 12.0.2
 - Hardware Platform: RPI4B

chewitt, 2025-01-28:
This was an intentional upstream change to prevent issues seen with two CEC devices active. You need to connect the device that needs CEC support to the HDMI port nearest the power connector (HDMI-A-1).

As this is not a bug .. closing the ticket.

closed 2025-01-28 by chewitt

jemiller00, 2025-01-28:
It might be worth nothing in the release notes. I just checked the website and wasn't able to find release notes for 12.0.2.