LibreELEC/LibreELEC.tv #9771: [BUG] Still cannot play back X265 streams due to mesa bugs?

oliv3r, 2025-02-06:
### Describe the bug
I was suffering from this issue on my old A10 (socket FM2 or some such) and filed a bug report on this years ago. https://github.com/LibreELEC/LibreELEC.tv/issues/7088

I accepted it as it was, a bug and figured it would get fixed after some updates. Alas, I bought a new AMD 8600g (socket am5, zen4?) and it still suffers from the same. So much for hoping new hardware would avoid the issue.

The annoying bit is, this most definitely worked in the past. Maybe during libreelec 11 times. I'll downgrade and see.

### How to reproduce
Steps to reproduce the behavior:
1. Go to video player
2. Play h265 encoded file
3. See error

### Information
 - LibreELEC Version: 12.02
 - Hardware Platform: amd64
### Log file
```
2025-02-06 21:37:03.231 T:1660     info <general>: VideoPlayer::OpenFile: ....1080p.WEBRip.x265-RARBG.mp4
2025-02-06 21:37:03.231 T:2243     info <general>: Creating InputStream
2025-02-06 21:37:03.788 T:2243     info <general>: Creating Demuxer
2025-02-06 21:37:03.866 T:2243     info <general>: Opening stream: 0 source: 256
2025-02-06 21:37:03.866 T:2243     info <general>: [WHITELIST] Searching the whitelist for: width: 1920, height: 960, fps: 23.976, 3D: false
2025-02-06 21:37:03.866 T:2243     info <general>: Display resolution ADJUST : 3840x2160 @ 23.976025 Hz (27) (weight: 0.000)
2025-02-06 21:37:03.872 T:1660     info <general>: VideoPlayer: OnLostDisplay received
2025-02-06 21:37:03.872 T:1660  warning <general>: CDVDMessageQueue(audio)::Put MSGQ_NOT_INITIALIZED
2025-02-06 21:37:03.872 T:1660  warning <general>: CDVDMessageQueue(video)::Put MSGQ_NOT_INITIALIZED
2025-02-06 21:37:03.872 T:1660     info <general>: GLES: Maximum texture width: 16384
2025-02-06 21:37:03.872 T:2243     info <general>: Creating video codec with codec id: 173
2025-02-06 21:37:03.872 T:2243     info <general>: CDVDVideoCodecDRMPRIME::Open - using decoder HEVC (High Efficiency Video Coding)
2025-02-06 21:37:03.874 T:2243     info <general>: Creating video thread
2025-02-06 21:37:03.874 T:2258     info <general>: running thread: video_thread
2025-02-06 21:37:03.874 T:2243     info <general>: Opening stream: 1 source: 256
2025-02-06 21:37:03.874 T:2243     info <general>: Finding audio codec for: 86018
2025-02-06 21:37:03.874 T:2243     info <general>: CDVDAudioCodecFFmpeg::Open() Successful opened audio decoder aac
2025-02-06 21:37:03.874 T:2243     info <general>: Creating audio thread
2025-02-06 21:37:03.874 T:2259     info <general>: running thread: CVideoPlayerAudio::Process()
2025-02-06 21:37:04.644 T:1660     info <general>: VideoPlayer: OnResetDisplay received
2025-02-06 21:37:04.645 T:2242     info <general>: CVideoSyncGbm::Setup: opened (fd:0 crtc:79 seq:41585 ns:695589831190:695487889452)
2025-02-06 21:37:04.676 T:2246    error <general>: CDVDVideoCodecDRMPRIME::GetFormat - no supported pixel formats: vaapi, yuv420p10le
2025-02-06 21:37:04.685 T:2259     info <general>: CVideoPlayerAudio: display reset occurred, checking for passthrough
2025-02-06 21:37:04.686 T:2259     info <general>: CDVDAudioCodecFFmpeg::Open() Successful opened audio decoder aac
2025-02-06 21:37:04.686 T:2259     info <general>: Creating audio stream (codec id: 86018, channels: 6, sample rate: 48000, no pass-through)
2025-02-06 21:37:04.708 T:2258    error <general>: CDVDVideoCodecDRMPRIME::AddData - send packet failed: Operation not permitted (-1)
2025-02-06 21:37:04.710 T:2258    error <general>: CDVDVideoCodecDRMPRIME::GetPicture - videoBuffer:nullptr format:yuv420p10le
2025-02-06 21:37:04.713 T:2258     info <general>: Skipped 8 duplicate messages..
2025-02-06 21:37:04.713 T:2258    error <general>: CDVDVideoCodecDRMPRIME::AddData - send packet failed: Invalid data found when processing input (-1094995529)
2025-02-06 21:37:04.714 T:2258    error <general>: CDVDVideoCodecDRMPRIME::GetPicture - videoBuffer:nullptr format:yuv420p10le
2025-02-06 21:37:04.847 T:1722     info <general>: Skipped 155 duplicate messages..
2025-02-06 21:37:04.847 T:1722     info <general>: CAESinkALSA - Unable to open device "default" for playback
2025-02-06 21:37:04.851 T:2258    error <general>: CDVDVideoCodecDRMPRIME::GetPicture - videoBuffer:nullptr format:yuv420p10le
2025-02-06 21:37:04.914 T:1722     info <general>: Skipped 13 duplicate messages..
```

heitbaum, 2025-02-06:
Pirate media

chewitt, 2025-02-07:
@oliv3r use Jellyfish files for testing and take discussion to the forum

chewitt, 2025-02-07:
Note to self https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/27730

oliv3r, 2025-02-07:
@chewitt what files now?

I just installed nightly but I think there's a few issues there; e.g.

```
nfs-ls: symbol lookup error: /usr/lib/libnfs.so.16: undefined symbol: gnutls_free
```

But are those worth to be reported even?

oliv3r, 2025-02-07:
https://forum.libreelec.tv/thread/29479-bug-still-cannot-play-back-x265-streams-due-to-mesa-bugs-9771