LibreELEC/LibreELEC.tv #9904: [RFE] migrate addons settings to new format

heitbaum, 2025-03-22:
Complete migration of settings from old to new
`trying to load setting definitions from old format`
- https://kodi.wiki/view/Add-on_settings
- https://kodi.wiki/view/Add-on_settings_conversion

e.g.
```
2025-03-23 08:44:57.560 T:2852963   debug <CAddonSettings[0@service.snapclient]>: trying to load setting definitions from old format...
2025-03-23 08:44:57.749 T:2852959   debug <CAddonSettings[0@service.librespot]>: trying to load setting definitions from old format...
2025-03-23 08:45:02.435 T:2852965   debug <CAddonSettings[0@service.system.docker]>: trying to load setting definitions from old format...
```

mglae, 2025-03-23:
We should do it step by step, e.g. while updating the addon.

---

It's already completed for:
```
packages/addons/service/locale/resources/settings.xml
packages/addons/service/ttyd/source/resources/settings.xml
```
Use `find packages/addons/ -type f -name settings.xml -print0|xargs -0 grep -l "settings version"`

---

An update is required for language file directories too (see https://forum.kodi.tv/showthread.php?tid=332789):
```
packages/addons/service/emby4/source/resources/language/English
packages/addons/service/tvheadend43/source/resources/language/English
packages/addons/service/jellyfin/source/resources/language/English
packages/addons/service/syncthing/source/resources/language/English
packages/addons/service/mpd/source/resources/language/English
packages/addons/service/oscam/source/resources/language/English
packages/addons/service/tvheadend42/source/resources/language/English
packages/addons/service/minidlna/source/resources/language/English
packages/addons/service/mariadb/source/resources/language/English
packages/addons/service/minisatip/source/resources/language/English
packages/addons/service/docker/source/resources/language/English
packages/addons/service/rsyslog/source/resources/language/English
packages/addons/service/lcdd/source/resources/language/English
packages/addons/service/vdr-addon/source/resources/language/English
packages/addons/service/snapserver/source/resources/language/English
packages/addons/service/nextpvr/source/resources/Language/English
packages/addons/service/prometheus-node-exporter/source/resources/language/English
packages/addons/service/net-snmp/source/resources/language/English
packages/addons/service/inadyn/source/resources/language/English
packages/addons/service/snapclient/source/resources/language/English
packages/addons/service/tigervnc/source/resources/language/English
packages/addons/service/bluetooth-audio/source/resources/language/English
packages/addons/service/tinc/source/resources/language/English
packages/addons/service/librespot/source/resources/language/English
packages/addons/service/filebrowser/source/resources/language/English
packages/addons/tools/ffmpeg-tools/source/resources/language/English
packages/addons/driver/hdhomerun/source/resources/language/English
packages/addons/browser/chrome/source/resources/language/English
```

heitbaum, 2025-03-23:
> We should do it step by step, e.g. while updating the addon.

I did Jellyfin this evening. A bit tedious, and did need some testing and rework. But agreed, we can merge them when the addons are being updated.

mglae, 2025-03-23:
Hopefully you did use the great [conversion tool](https://forum.kodi.tv/showthread.php?tid=350135)

heitbaum, 2026-01-16:
> > [@emveepee](https://github.com/emveepee) all the addons have now been updated to the new style Language and new Style settings.xml. I haven’t reviewed the NextPVR settings line by line, but did note that the first line is the old style (although version 1 is set.) From the wiki it should be `<?xml version="1.0" ?>`
> 
> Thanks [@heitbaum](https://github.com/heitbaum) I am prepping a PR for version 7.0.5 anyway and will look and change what is necessary Any issue with backporting this to LE 11. The styles aren't new for Piers and you indicated you wanted to to keep my PR's aligned.

I have backported all these to 12.2 - so all good. Was since Kodi 19 Matrix (some enhancements were added in v20)
LE11? Hasn’t seen any recent builds? CI doesn’t build for LE11 or LE12 anymore.