LibreELEC/LibreELEC.tv #9916: Prepare LE to build with cmake 4.0.0

heitbaum, 2025-03-28:
Upstream repo already in repo
-  libftdi1: patches since 1.5 allowing build with cmake-4.0.0
 - http://developer.intra2net.com/git/?p=libftdi
 - patched with "since 1.5"
- libsndfile: allow build with cmake-4.0.0
  - from upstream
- libarchive: allow build with cmake-4.0.0
  - from upstream
- libevent: allow build with cmake-4.0.0
  - https://github.com/libevent/libevent/issues/1782
  - Included in the master codebase 

Upstream patches raised
- kodi-platform: allow build with cmake-4.0.0
  - https://github.com/xbmc/kodi-platform/pull/31
- media-driver: allow build with cmake-4.0.0
  - https://github.com/intel/media-driver/issues/1918
- gmmlib: allow build with cmake-4.0.0
  - https://github.com/intel/gmmlib/issues/129
- rapidjson: allow build with cmake-4.0.0
  - https://github.com/Tencent/rapidjson/issues/2333
- libogg: allow build with cmake-4.0.0
  - https://gitlab.xiph.org/xiph/ogg/-/issues/2304
- lzo: allow build with cmake-4.0.0
  - email sent to oberhumer

Local patch
- p8-platform: allow build with cmake-4.0.0
  - unmaintained 
- soxr: allow build with cmake-4.0.0
  - https://sourceforge.net/p/soxr/code/ci/master/tree/

fuzzard, 2025-03-28:
You may or may not be aware of https://cmake.org/cmake/help/latest/variable/CMAKE_POLICY_VERSION_MINIMUM.html

It allows to set a "minimum" that will override what a cmake project has its min version at
Adding a define to a configure command will then "avoid" the cmake 4 failure

```
-DCMAKE_POLICY_VERSION_MINIMUM=3.5
```

Essentially avoiding having to patch for projects that arent actively maintained (lzo2 is a prime example from kodi perspective).

Cmake state the following in their pre release notes about the new variable

```
The “CMAKE_POLICY_VERSION_MINIMUM” variable was added to help
packagers and end users try to configure existing projects that have
not been updated to work with supported CMake versions.
```

Do be aware, cmake devs have stated they intend to make further breaking changes from the 3.x branch (deprecated removals), so setting the minimum to 3.5 in the example above, may need to be revisited in future cmake 4 version bumps if they drop deprecated features in > 3.5

heitbaum, 2025-03-28:
> You may or may not be aware of https://cmake.org/cmake/help/latest/variable/CMAKE_POLICY_VERSION_MINIMUM.html
>
> It allows to set a "minimum" that will override what a cmake project has its min [...]

I did think about it. I’m regretting not patching some of the gcc-12,13,14… workarounds I had made when we bumped previously, as remembering to clean them up / or revisiting the CFLAGS. So have been on the upstream patch bandwagon for both the gcc15 and now the cmake4. 

Some of the patches I have written are 3.10 targeted, which seems to be the not yet deprecated version. When there has been inconsistencies in the version (aka gmmlib and media-driver) then I have just been updating to the consistent.

heitbaum, 2025-03-28:
> All good. Internally in kodi for other platforms i'll just be using the override for things like lzo.
> 
> Just wanted to make sure it was known about a way so you dont have to patch things that are abandoned (or less maintained) upstream and then carry more patches for good

You are right. Once we get down to the patches that are/will be accepted, then agree will be time to use the `CMAKE_POLICY_VERSION_MINIMUM` override

heitbaum, 2025-03-29:
@fuzzard - 29 patches required - so not to bad, pretty much all upstreamed, most were simple patches, a couple required rework of the CMakeLists.txt. Let’s see how much traction we get on the upstream merges.

PR #9918 (2025-04-18): CMake update to 4.x
~This is a draft tracking pr with all the required patches/changes (except hyperhdr) to build LE and addons with cmake-4.0.0 on x86_64. Some of these patches will be separated out to be merged directly (especially the local patches for the unmaintained / local fixes.)~

~The build is wip to test build on other targets.~

~It is expected that some of these patches will be incorporated in package updates, depending on upstream release schedules.~

- Fixes #9916
- first group of changes #9937
- Second group of change #9964
closed 2025-04-18 by chewitt