LibreELEC/LibreELEC.tv #9951: [BUG] Pi netboot - 12.0.2 mounts NFS /storage as read-only

DavidCWGA, 2025-04-11:
When booting LibreELEC 12.0.2 on a Pi (5, if it matters) using NFS netboot, the /storage mount is mounted "ro", which breaks everything.

I added "textmode" to cmdline.txt and then did `mount -o remount,rw /storage`, which worked fine, which seems to rule out a server issue.

My cmdline.txt looks like this:
`boot=NFS=<NFS server IP>:/path/to/boot disk=NFS=<NFS server IP>:/path/to/disk ssh ip=dhcp quiet console=ttyAMA10,115200 console=tty0`

This is broken in 12.0.2, but works OK in 12.0.1.

closed 2025-04-11 by heitbaum

DavidCWGA, 2025-04-12:
I retested. It works in 12.0.1,
is broken in 12.0.2,
is broken in 12.0-nightly-20250404-5c24d2b.
and is broken in 13.0-nightly-20250410-fe6b62d

I do not have permission to reopen this issue, "completed" would seem to be a mistake.

heitbaum, 2025-04-12:
Best to raise a forum post. The team to help diagnose will look at the post there.