LibreELEC/LibreELEC.tv #9983: [BUG] [LE13] nightly 20250418-554cfb7not seen in the GUI-Updater

eldersixpack13, 2025-04-19:
### Describe the bug
nightly 20250418-554cfb7not seen in the GUI-Updater, but could be fetch via an script [1]
it seems Updates stucks somehow on 20250417-99b0905
tested tonight ~23:00 and today ~13:00

### How to reproduce
Steps to reproduce the behavior:
1. open the GUI Updater

### Information
 - LibreELEC Version:LE13
 - Hardware Platform: NUC13

### Log file
photo

[image]

### Context

[1]
script:
https://gist.github.com/eldersixpack13

eldersixpack13, 2025-04-19:
fixed

eldersixpack13, 2025-04-20:
@eldersixpack13, you're an idiot !

yes Sir.
Thank you Sir !
;-)

===
it's still an bug (here alone ?)

https://test.libreelec.tv/releases.json contains:
- LibreELEC-Generic.x86_64-13.0-nightly-20250417-99b0905.img.gz
- "timestamp": "2025-04-17 06:04:32"

and nothing newer (at least for generic)

CvH, 2025-04-20:
https://github.com/LibreELEC/actions/actions/runs/14563377945/job/40849457334
fixed

job was disabled because nothing was commited to the action repo for 2 months

closed 2025-04-20 by CvH