LibreELEC/LibreELEC.tv #10027: [BUG] LibreELEC 12 BananaPi BPI-M5

DavidHenryThoreau, 2025-05-01:
### Describe the bug

I've installed the latest stable version for Banana Pi BPM-5 
https://releases.libreelec.tv/LibreELEC-AMLGX.aarch64-12.0.2-bananapi-m5.img.gz
On two different sdcard. I've got that error : 
`Error in mount_flash could not mount LABEL=LIBREELEC`
I've tried to installed libreelec image by using : 
`dd if=LibreELEC-AMLGX.aarch64-12.0.2-bananapi-m5.img of=/dev/sd status=progress bs=4M`
and [balenaEtcher](https://etcher.balena.io/) on the two sdcards.
I've also mount the sdcard untill install is finished and changed :

```
  APPEND boot=LABEL=LIBREELEC disk=LABEL=STORAGE quiet systemd.debug_shell=ttyAML0 console=ttyAML0,115200n8 console=tty0

to

  APPEND boot=UUID=1901-5155 disk=UUID=5d100809-b774-4517-8bd3-75629a1f323c quiet ssh%
```  

### How to reproduce
Steps to reproduce the behavior:
1. Install the image to sdcards => different size 32Gio and 256Gio
2. The error appears on screen 
3. `Error in mount_flash could not mount LABEL=LIBREELEC`

### Information
 - LibreELEC Version: 12.0.2
 - Hardware Platform: Banana-Pi BPI-m5
  
 ### Log file

### Context

DavidHenryThoreau, 2025-05-01:
I've also tried others images after check sha256sum which is ok : 
https://test.libreelec.tv/12.0/Amlogic/bananapi-m5/LibreELEC-AMLGX.aarch64-12.0-nightly-20250416-e25b5e0-bananapi-m5.img.gz 
and old one :  
https://releases.libreelec.tv/LibreELEC-AMLGX.arm-11.0.6-bananapi-m5.img.gz

Still the same error : 
https://forum.libreelec.tv/core/attachment/2202-minix-u9-h-error-png/

DavidHenryThoreau, 2025-05-01:
Also tried : 
LibreELEC-AMLGX.aarch64-13.0-nightly-20250430-b24a5d7-bananapi-m5.img
Still the same

chewitt, 2025-05-02:
@DavidHenryThoreau please test https://chewitt.libreelec.tv/testing/LibreELEC-AMLGX.aarch64-12.80.2-bananapi-m5.img.gz to see if it gives the same result, the kernel has some differences to the one in current LE13 nightlies.

If not booting still it would be helpful to see the UART output from the board. Having the UART console connected will allow you to poke around in the initramfs boot environment to look for kernel issues and attempt manual mounting (to see what errors are generated). Unfortunately the M5 board that I have appears to have died since I last booted it (about a year ago) so I'm unable to check anything myself.

NB: I am 100% sure Minix U9-H issues have a different root cause although not mounting the boot volume and the generic failure boot message that you then see are the same.

https://github.com/libretro/Lakka-LibreELEC/issues/2059 is probably the same issue.

DavidHenryThoreau, 2025-05-02:
@chewitt Thanks for your answer ! 
I've installed your build : 
```
 sudo dd if=LibreELEC-AMLGX.aarch64-12.80.2-bananapi-m5.img of=/dev/mmcblk0 bs=4M status=progress
```
But i've still the same message.

Can you say me more with UART, do you have a documentation about it ?

> https://github.com/libretro/Lakka-LibreELEC/issues/2059 is probably the same issue.
Yes , I've seen that issue too.

DavidHenryThoreau, 2025-05-02:
I've got a USBtoTTL like this one : 
https://i.ebayimg.com/images/g/95oAAOSwJnJj4L0z/s-l1600.jpg
I'm trying to find if BananaPi requires a 3V or 5V.
https://wiki.banana-pi.org/Banana_Pi_BPI-M5

Does banana need 3V or 5V ?

chewitt, 2025-05-02:
Only the GND and TX/RX pins are required (no need to connect the 3/5V pin). There's a header behind the USB ports.

DavidHenryThoreau, 2025-05-02:
```
SM1:BL:511f6b:81ca2f;FEAT:A0F83180:20282000;POC:F;RCY:0;EMMC:0;READ:0;CHK:1F;READ:0;CHK:1F;READ:0;CHK:1F;SD?:0;SD:0;READ:0;0.0;CHK:0;
bl2_stage_init 0x01
bl2_stage_init 0x81
hw id: 0x0000 - pwm id 0x01
bl2_stage_init 0xc1
bl2_stage_init 0x02

no sdio debug board detected 
L0:00000000
L1:00000703
L2:00008067
L3:15000020
S1:00000000
B2:20282000
B1:a0f83180

TE: 287894

BL2 Built : 15:21:42, Mar 26 2020. g12a g486bc38 - gongwei.chen@droid11-sz

Board ID = 1
Set cpu clk to 24M
Set clk81 to 24M
Use GP1_pll as DSU clk.
DSU clk: 1200 Mhz
CPU clk: 1200 MHz
Set clk81 to 166.6M
board id: 1
Load FIP HDR DDR from SD, src: 0x00010200, des: 0xfffd0000, size: 0x00004000, part: 0
fw parse done
PIEI prepare done
fastboot data verify
result: 255
Cfg max: 12, cur: 1. Board id: 255. Force loop cfg
LPDDR4 probe

LPDDR4_PHY_V_0_1_21-Built : 15:37:51, Mar 26 2020. g12a gf098346 - gongwei.chen@droid11-sz
ddr clk to 1392MHz

dmc_version 0001
Check phy result
INFO : End of CA training
INFO : End of initialization
INFO : Training has run successfully!
Check phy result
INFO : End of initialization
INFO : End of read enable training
INFO : End of fine write leveling
INFO : End of Write leveling coarse delay
INFO : Training has run successfully!
Check phy result
INFO : End of initialization
INFO : End of read dq deskew training
INFO : End of MPR read delay center optimization
INFO : End of write delay center optimization
INFO : End of read delay center optimization
INFO : End of max read latency training
INFO : Training has run successfully!
1D training succeed
Check phy result
INFO : End of initialization
INFO : End of 2D read delay Voltage center optimization
INFO : End of 2D read delay Voltage center optimization
INFO : End of 2D write delay Voltage center optimization
INFO : End of 2D write delay Voltage center optimization
INFO : Training has run successfully!

soc_vref_reg_value 0x 0000001d 0000001d 0000001c 0000001d 0000001d 0000001c 0000001d 0000001c 0000001d 0000001c 0000001f 0000001e 0000001d 0000001b 0000001b 0000001c 0000001e 0000001d 0000001e 0000001d 0000001e 0000001d 0000001d 0000001d 0000001e 0000001e 0000001e 0000001f 0000001e 0000001e 00000020 0000001d dwc_ddrphy_apb_wr((0<<20)|(2<<16)|(0<<12)|(0xb0):0004 
 dram_vref_reg_value 0x 0000004c
2D training succeed
auto size-- 65535DDR cs0 size: 2048MB
DDR cs1 size: 2048MB
DMC_DDR_CTRL: 00e00024DDR size: 3928MB
cs0 DataBus test pass
cs1 DataBus test pass
cs0 AddrBus test pass
cs1 AddrBus test pass

non-sec scramble use zero key
ddr scramble enabled

100bdlr_step_size ps== 399
result report
boot times 0Enable ddr reg access
Load FIP HDR from SD, src: 0x00010200, des: 0x01700000, size: 0x00004000, part: 0
Load BL3X from SD, src: 0x00078200, des: 0x01768000, size: 0x000f5000, part: 0
0.0;M3 CHK:0;cm4_sp_mode 0
MVN_1=0x00000000
MVN_2=0x00000000
[Image: g12a_v1.1.3386-3b31431 2019-05-21 10:41:54 luan.yuan@droid15-sz]
OPS=0x10
ring efuse init
2b 0c 10 00 01 1c 0c 00 00 15 32 31 36 58 4b 50 
[0.017319 Inits done]
secure task start!
high task start!
low task start!
boot bl31
NOTICE:  BL31: v1.3(release):4fc40b1
NOTICE:  BL31: Built : 15:57:33, May 22 2019
NOTICE:  BL31: G12A normal boot!
ERROR:   Error initializing runtime service opteed_fast

<debug_uart>

U-Boot 2025.04 (Apr 30 2025 - 12:03:49 +0000)bpi-m5

Model: Banana Pi BPI-M5
SoC:   Amlogic Meson SM1 (S905X3) Revision 2b:c (10:2)
DRAM:  1 GiB (effective 3.8 GiB)
Core:  396 devices, 30 uclasses, devicetree: separate
MMC:   mmc@ffe05000: 0, mmc@ffe07000: 1
Loading Environment from nowhere... OK
In:    usbkbd,serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@ff3f0000
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found /extlinux/extlinux.conf
Retrieving file: /extlinux/extlinux.conf
1:	LibreELEC
Retrieving file: /KERNEL
append: boot=LABEL=LIBREELEC disk=LABEL=STORAGE quiet systemd.debug_shell=ttyAML0 console=ttyAML0,115200n8 console=tty0
Retrieving file: /amlogic/meson-sm1-bananapi-m5.dtb
## Booting kernel from Legacy Image at 08080000 ...
   Image Name:   
   Image Type:   AArch64 Linux Kernel Image (lzo compressed)
   Data Size:    12887575 Bytes = 12.3 MiB
   Load Address: 01d80000
   Entry Point:  01d80000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 08008000
   Booting using the fdt blob at 0x8008000
Working FDT set to 8008000
   Uncompressing Kernel Image to 1d80000
Moving Image from 0x1d80000 to 0x1e00000, end=0x38c0000
   Loading Device Tree to 000000003ffea000, end 000000003ffffdd8 ... OK
Working FDT set to 3ffea000

Starting kernel ...

[    0.000000] RME: RMM doesn't support RSI version 1.0. Supported range: 1.0-0.0
[    0.375396] meson-drm ff900000.vpu: DSI transceiver device is disabled
```

DavidHenryThoreau, 2025-05-02:
Thanks for quick feedback 👍 
Here's the output by using `earlycon`
https://pastebin.com/raw/W8RXyPjJ

chewitt, 2025-05-03:
Remove `quiet` from boot params? .. I'm trying to see more output (to see if there are error messages).

DavidHenryThoreau, 2025-05-03:
I've removed `quiet`
https://pastebin.com/raw/tywNZFr0

DavidHenryThoreau, 2025-05-03:
I see that unknown command line parameters
```
[    0.000000] Unknown kernel command line parameters "boot=LABEL=LIBREELEC disk=LABEL=STORAGE", will be passed to user space.
```

chewitt, 2025-05-03:
The `Unknown kernel command line parameters` warning is harmless. What happens if you attempt to mount partitions?

```
mkdir -p /flash
mkdir -p /storage
mount /dev/mmcblk0p1 /flash
mount /dev/mmcblk0p2 /storage
```

If they fail to mount, fsck the partitions and try again. If they mount, type `boot` and see what happens.

DavidHenryThoreau, 2025-05-03:
```
mount /dev/mmcblk0 /flash
mount: mounting /dev/mmcblk0 on flash failed: invalid argument

mount /dev/mmcblk1 /storage
mount: mounting /dev/mmcblk1 on storage failed: invalid argument
```

chewitt, 2025-05-03:
You've attempted to mount the root of two mmc devices not the first/second partitions of the mmcblk0 device.

User error :)

Do `mount /dev/mmcblk0p1 /flash` not `mount /dev/mmcblk0 /flash` etc.

DavidHenryThoreau, 2025-05-03:
I don't have `mmcblk0p1` or `mmcblk0p1`

I only have : 
```
mount /dev/mmcblk
mmcblk0 mmcblk1 mmcblk1boot0 mmcblk1boot1 mmcblk1rpmb
```

DavidHenryThoreau, 2025-05-03:
I don't have `mmcblk0p1` or `mmcblk0p2`

I only have : 
```
mount /dev/mmcblk
mmcblk0 mmcblk1 mmcblk1boot0 mmcblk1boot1 mmcblk1rpmb
```

chewitt, 2025-05-03:
Most odd. The board has booted u-boot from SD which loads init from KERNEL from the first partition of the SD card; but the kernel then fails to mount /flash (and /storage) as /dev/mmcblk0p1/2 (the partitions on the card) don't appear to have probed fully and thus don't exist. The system log does show intial detection of the mmcblk0 device, but nothing more.

What's the spec of the PSU are you using? I've seen unexplainable SD/EMMC problems on M5 and M2S boards before due to the board not being able to draw enough power.

Can you also try writing the LE image to a USB stick instead of an SD card. The board should be able to boot from USB too.

DavidHenryThoreau, 2025-05-03:
I've bought a BananaPi Kit ( the power supply output is : 
3.6-6.5V 3A; 6.5-9V 2A; 9-12V 1.5A

I've tried USB stick instead, nothing appears.

DavidHenryThoreau, 2025-05-03:
I've bought a BananaPi Kit, the power supply output is : 
3.6-6.5V 3A; 6.5-9V 2A; 9-12V 1.5A

I've tried USB stick instead, nothing appears.

chewitt, 2025-05-03:
Please try the USB stick in all ports, in case the board will only boot from the OTG port.

If no change, download and write this to an SD card: https://chewitt.libreelec.tv/testing/LibreELEC-AMLGX.aarch64-12.80.2-bananapi-m5.img.gz - I've reduced the SD card speed in device-tree to see if that improves stability.

DavidHenryThoreau, 2025-05-03:
LibreELEC on USB stick does not start in any ports.

This image boots ! 👍 I've tried to switch menus preferences and it seems good ! Nice jobs ! 😄

chewitt, 2025-05-03:
The same change (dropping from 50MHz to 35MHz) is already done for the ac2xx dtsi. I'll get the Lakka folks to test the same change, and if all is good for them too I'll send the patch upstream.

DavidHenryThoreau, 2025-05-03:
- [ok] install the system to EMMC 
- [ok] network
- [ok] keyboard layouts
- [ok] ssh 

For me you can send patch, I'll try some days until it'll be merged to master.
Thanks for your help, I can now use my BananaPi ! 😃

DavidHenryThoreau, 2025-05-04:
I can't play some videos, here's `kodi.log` output error : 
```
error <general>: CDVDVideoCodecDRMPRIME::GetFormat - no supported pixel formats: yuv420p10le
info <general>: CAESinkALSA::InitializeHW - Your hardware does not support AE_FMT_FLOAT, trying other formats
 error <general>: CDVDVideoCodecDRMPRIME::AddData - send packet failed: Operation not permitted (-1)

```

DavidHenryThoreau, 2025-05-04:
I've used LibreELEC backup in order to backup RPI3 and restore that import to bananaPi M5

chewitt, 2025-05-04:
Yup, upstream support for Amlogic hardware decoding sucks balls, esp. on g12a/g12b/sm1 hardware, and the images in my share have some patches related to some experiments I was doing recently which may make things worse. I suggest you read any recent release notes on the website to understand the general state of Amlogic support.

DavidHenryThoreau, 2025-05-04:
Right, I'm gonna wait for final stable version.

fivemtebex, 2025-05-10:
Interesting solution to this problem! I've been working on similar challenges recently.

fivemserverhostings, 2025-05-10:
Your approach to this problem is very elegant. I'm learning a lot from reviewing your code.

fivemeyestore, 2025-05-10:
The documentation in this repository is excellent! Very clear and comprehensive.

fivemhostingserver, 2025-05-10:
Interesting solution to this problem! I've been working on similar challenges recently.

scriptforfivem, 2025-05-10:
Great implementation of this feature. I particularly like the approach to error handling.

chewitt, 2025-05-11:
Fixed in #10050

closed 2025-05-11 by chewitt