LibreELEC/LibreELEC.tv #10073: [BUG] [LE13] gbmshell is broken

eldersixpack13, 2025-05-19:
### Describe the bug
this helpfull tool doesn't work anymore:
https://forum.libreelec.tv/thread/28187-gbmshell-development-addon/

### How to reproduce
Steps to reproduce the behavior:
1. run gbmshell

### Information
 - LibreELEC Version: LE13 (last nightly)
 - Hardware Platform: NUC13

### Log file

### Context
as I do not have a account in the LE Forum I hope @mglae gets a notice.

To me the bug sems to be the change of the VAR'S in /etc/os-release 
LIBREEELEC_*  =>  DISTRO_*

closed 2025-05-19 by chewitt

eldersixpack13, 2025-05-19:
for the record (and/or if someone runs into this bug too)

changing the VAR in
/storage/.kodi/addons/tools.gbmshell/default.py

form 
if osrelease['LIBREELEC_ARCH'] ...
to 
if osrelease['DISTRO_ARCH'] not in non_gbm[version_id]:

fixes the bug

mglae, 2025-05-21:
Fixed in 1.0.3