LibreELEC/LibreELEC.tv #10117: [BUG] Display 3D as 2D in Generic (x86_X64) [LE13]

Voosjey, 2025-06-04:
### Describe the bug

We now have LE13 in the nightly builds, and you still cannot watch 3D movies in 2D on the Generic x86_x64. Do we all now have to convert our 3D movies into 2D so that we can watch them again? It must be possible to split an HSBS movie, especially since the 2D view is still integrated, and when you turn off 2D, it also doubles the screen, and when you then re-enable 2D, you only have the two images again. So there seems to be some function behind the 2D. It just doesn't work properly. Please check again. 

### How to reproduce
Steps to reproduce the behavior:
That's how the movie starts
[image]
Then you turn on the 2D view, and nothing happens, the double image remains. 
If you now turn off the 2D, you suddenly have 4 images side by side.
[image]
If you turn the 2D back on now, you only have both images again.
[image]

### Information
 - LibreELEC Version: [LE13 / LE12 / LE11]
 - Hardware Platform: [Intel i512400 Gigabyte Bord]

### Log file

No errors are displayed.

### Context

Voosjey, 2025-06-04:
The discussion is about a Raspberry Pi, not about an Intel system, and even if it were the same problem, it is still not worked and so is that a BUG. Where can I report the problem so that it is addressed and not just closed?

heitbaum, 2025-06-04:
> The discussion is about a Raspberry Pi, not about an Intel system, and even if it were the same problem, it is still not worked and so is that a BUG. Where can I report the problem so that it is addressed and not just closed?

The forum is the best place to start.