LibreELEC/LibreELEC.tv #10393: [BUG] Pi5 Regression: HDR not detected with Samsung S90D TV

BreakPoints, 2025-08-22:
Since upgrading to LibreELEC 12.2.0 from 12.0.2, Kodi no longer detects any of the HDR modes of my Samsung S90D TV (running latest firmware). In 12.0.2 the "Display supported HDR types" system info field correctly shows "HDR10, HLG" whereas in 12.2.0 it shows as "none".

I am unsure if this is limited to my Samsung as I only have 1 HDR compatible TV.

Downgrading back to 12.0.2 fixes the issue and I am again able to watch HDR videos in HDR mode.

### How to reproduce
Steps to reproduce the behavior:
1. Run LibreELEC 12.2.0 on a Pi5 and HDR TV.
2. Check System info -> Video -> Display supported HDR types **or** try playing an HDR formatted video.

### Information
 - LibreELEC Version: 12.2.0
 - Hardware Platform: RPi5
 - Display: Samsung S90D (QE65S90DATXXU)

adriankwil, 2025-12-07:
+1
I am seeing this exact same behaviour.
Disabling Input Signal Plus appears to make the pi see the tv as HDR compatible, which is a bit weird because in the samsung settings it says that HDR requires ISP enabled.

chewitt, 2025-12-07:
When there's a stable release of Kodi Piers