LibreELEC/LibreELEC.tv #10442: [BUG] 10-bit encoded HEVC files completely garbled on Rock64 (rk3328)

diederikdehaas, 2025-09-02:
### Describe the bug
When I try to play a 10-bit encoded HEVC/x265 file, its output is completely garbled. Surprisingly (to me ... see 'Context') it does play 10-bit encoded x264 files fine and looking at CPU utilization on htop, that appears to be HW decoded.

My conv2x264 script: https://paste.sr.ht/~diederik/32f057d138e46710b4bd956dca42bb5170bb689b
My conv2x265 script: https://paste.sr.ht/~diederik/7fca4cd2d4a85569faf8bb263b2cfe43c3691dcb

### How to reproduce
Steps to reproduce the behavior:
1. Get a standard big buck bunny video
2. Convert it to a 10-bit encoded HEVC file with the above linked 'conv2x265' script
3. Play it on LibreELEC on a Rock64 (it may manifest on various devices, but thus far only tested it on Rock64)
4. See the output completely garbled

In a reply to this issue I'll attach a photo of that garbled output, but attaching files currently fails here (probably me blocking some script, but I don't want to lose this bug report trying to fix/unblock that)

### Information
 - LibreELEC Version: 12.2.0 (RK3328.aarch64)
 - Hardware Platform: Rock64 (rk3328)

### Log file
I haven't checked whether there is an error in the log; I suspect there isn't.

### Context
I participate (by testing) in upstreaming efforts wrt HW decoding and V4L2request on Rockchip devices and for that I have several test video files. And I'm encountering a LOT of problems with 10-bit encoded files.
But that is with a [patched ffmpeg](https://github.com/Kwiboo/FFmpeg/tree/v4l2request-2025-v3-rkvdec) and with the build artifacts of that (ie the -dev packages), I build a [patched mpv](https://github.com/mpv-player/mpv/pull/14690). I then try to play my test video files with mpv running in sway (1.11) on a self-built 6.17-rcX kernel with patches by 'Kwiboo' and 'detlev' for HW based decoding.

I have problems with all my 10-bit encoded test files (thus both x264 and x265) and I am/was frustrated that it seems I am the only one who sees this, while it is 100% easily reproducible. So while I realize that LE uses a somewhat different 'display pipeline' (with GBM), I figured "what the heck, lets see if LE has any problems as well" ... and this issue shows there is indeed one.

FWIW: the problem that I'm running into with mpv on sway is that 10-bit encoded files are completely blue.
And I see these errors in the console when I try to play them (with v4l2request hw accel):
```
[ffmpeg] V4L2RequestContext: Failed to set 1 control(s): Invalid argument (22)
Using hardware decoding (v4l2request).
AO: [pipewire] 48000Hz stereo 2ch floatp
VO: [gpu] 1280x640 drm_prime[nv15]
[vo/gpu] Initializing texture for hardware decoding failed.
```

knaerzche, 2025-09-02:
Hi @diederikdehaas thanks for tracking that issue. 
I'm not sure I can fully follow why and what you did. Also I'm not sure what exactly ffmpeg does when "converting" an 8 bit stream to 10 bit - but certainly the additional 2 bit will not magically appear from "somewhere". It's also unlikely that 10-bit support is generally broken as the implementation has been tested several (lot) times with both h.264 and h.265 reference tests.
Can you upload the file somewhere? (some seconds are enough)

diederikdehaas, 2025-09-03:
I already mentioned this on IRC, but should mention it here as well:
I tested an image with the patch @Kwiboo mentioned integrated and the problem was fixed. Thanks!
While it was version 12.90.1, I could reproduce the original problem on a previous 12.90.1 build and as the change was related to a change in the 6.16, which both have, I think it's safe to assume that fix will fix the issue on 12.2 as well.

XxAcielxX, 2025-09-08:
LibreELEC (official): 12.2.0 (RK3399.aarch64)
Board: NanoPC-T4

Is also affected by the same issue too. Hope for a speedy fix as I'm unable to watch my media which in majority consists of HEVC.

chewitt, 2025-09-08:
Fixed in https://github.com/LibreELEC/LibreELEC.tv/pull/10450

I'm sure there are further issues to fix, but those need to be isolated and have specific issues raised rather than have catch-all sounding reports that attract #metoo posts from users.

XxAcielxX, 2025-09-10:
@chewitt Can you kindly let us know which Nightly build (12.2) this fix will be available from?

chewitt, 2025-09-10:
I suck at spoon feeding the obvious to people so you might need to use an ounce of initiative. I guarantee the fix is not present in any image dated before the fix was merged. There are probably some images since then.