LibreELEC/LibreELEC.tv #10532: [BUG] Error messages to find and fix

chewitt, 2025-09-30:
The following error messages are observed during boot. All appear to be completely harmless, but clean logs are prettier :)

```
Sep 17 16:00:40 N2PLUS systemd-udevd[452]: /usr/lib/udev/rules.d/50-udev-default.rules:13 Unknown group 'clock', ignoring.
Sep 17 16:00:40 N2PLUS systemd-udevd[452]: /usr/lib/udev/rules.d/50-udev-default.rules:34 Unknown group 'clock', ignoring.
```

```
Sep 17 16:00:41 N2PLUS (udev-worker)[508]: controlC0: /usr/lib/udev/rules.d/90-alsa-restore.rules:5 NAME="snd/%k": Only network interfaces can be renamed, ignoring.
```

```
Sep 29 17:18:56 N2PLUS pulseaudio[467]: W: [pulseaudio] authkey.c: Failed to open cookie file '/var/run/pulse/.config/pulse/cookie': No such file or directory
Sep 29 17:18:56 N2PLUS pulseaudio[467]: W: [pulseaudio] authkey.c: Failed to load authentication key '/var/run/pulse/.config/pulse/cookie': No such file or directory
Sep 29 17:18:56 N2PLUS pulseaudio[467]: W: [pulseaudio] authkey.c: Failed to open cookie file '/var/run/pulse/.pulse-cookie': No such file or directory
Sep 29 17:18:56 N2PLUS pulseaudio[467]: W: [pulseaudio] authkey.c: Failed to load authentication key '/var/run/pulse/.pulse-cookie': No such file or directory
```

```
Sep 29 17:18:56 N2PLUS pulseaudio[467]: E: [pulseaudio] bluez5-util.c: GetManagedObjects() failed: org.freedesktop.systemd1.NoSuchUnit: Unit dbus-org.bluez.service not found.
```

chewitt, 2026-03-04:
This is also a long-time annoyance in Kodi logs; again the fix is probably adapting Kodi to our packaging rather than adding something missing or wrong in LE:

```
2026-03-04 13:12:02.892 T:878     debug <general>: DBus method call to org.freedesktop.ConsoleKit.Manager.CanStop at /org/freedesktop/ConsoleKit/Manager of org.freedesktop.ConsoleKit failed: org.freedesktop.DBus.Error.ServiceUnknown - The name org.freedesktop.ConsoleKit was not provided by any .service files
2026-03-04 13:12:02.892 T:878      info <general>: Selected Logind/UPower as PowerSyscall
2026-03-04 13:12:02.892 T:878     error <general>: DBus error: org.freedesktop.DBus.Error.ServiceUnknown - The name org.freedesktop.UPower was not provided by any .service files
2026-03-04 13:12:02.892 T:878      info <general>: LogindUPowerSyscall - UPower not found, battery information will not be available
2026-03-04 13:12:02.906 T:878     debug <general>: LogindUPowerSyscall - inhibit lock taken, fd 12
2026-03-04 13:12:02.907 T:878     debug <general>: DBus method call to org.freedesktop.UPower.EnumerateDevices at /org/freedesktop/UPower of org.freedesktop.UPower failed: org.freedesktop.DBus.Error.ServiceUnknown - The name org.freedesktop.UPower was not provided by any .service files
2026-03-04 13:12:02.907 T:878     debug <general>: DBus method call to org.freedesktop.DBus.Peer.Ping at /org/freedesktop/UDisks2 of org.freedesktop.UDisks2 failed: org.freedesktop.DBus.Error.ServiceUnknown - The name org.freedesktop.UDisks2 was not provided by any .service files
2026-03-04 13:12:02.908 T:878     debug <general>: DBus method call to org.freedesktop.UDisks.EnumerateDevices at /org/freedesktop/UDisks of org.freedesktop.UDisks failed: org.freedesktop.DBus.Error.ServiceUnknown - The name org.freedesktop.UDisks was not provided by any .service files
2026-03-04 13:12:02.908 T:878     debug <general>: Selected UDev as storage provider
```

HiassofT, 2026-03-05:
> ```
> 2026-03-04 13:12:03.575 T:878     error <general>: [xkb] [XKB-679] No Compose file for locale "en_US.UTF-8": locale is either invalid or not installed
> 2026-03-04 13:12:03.575 T:878     error <general>: [xkb] [XKB-679] couldn't find a Compose file for locale "C" (mapped to "C")
> 2026-03-04 13:12:03.575 T:878   warning <general>: CLibInputKeyboard::CLibInputKeyboard(): Failed to compile localized compose table, composed key support will be disabled
> ```

The easiest way to silence this messages is to point the `XCOMPOSE` env var to an almost empty dummy file, eg with contents `# empty` (empty file will result in an error message).

This disables compose and dead-key handling (which we don't need in LE IMO).

Alternative would be to install the `/usr/share/X11/locale/` stuff from the libX11 package - composition is locale dependent

heitbaum, 2026-10-07:
> Another one:
> 
> ```
> Sep 17 16:00:40 LibreELEC systemd-tmpfiles[365]: Failed to create directory or subvolume "/root", ignoring: Read-only file system
> Sep 17 16:00:40 LibreELEC systemd-tmpfiles[365]: Failed to open path '/root', ignoring: No such file or directory
> ```

- fixed by #11806

heitbaum, 2026-10-07:
> Adding to the list, this is in Kodi but smells like LE packaging differing from what's expected; fix could be either on our side or perhaps adapting Kodi to deal with our difference:
> 
> ```
> 2026-03-04 13:12:03.575 T:878     error <general>: [xkb] [XKB-679] No Compose file for locale "en_US.UTF-8": locale is either invalid or not installed
> 2026-03-04 13:12:03.575 T:878     error <general>: [xkb] [XKB-679] couldn't find a Compose file for locale "C" (mapped to "C")
> 2026-03-04 13:12:03.575 T:878   warning <general>: CLibInputKeyboard::CLibInputKeyboard(): Failed to compile localized compose table, composed key support will be disabled
> ```

- fixed in #11883

heitbaum, 2026-10-07:
Audit:

I tested Generic on AMD Ser7, Rock Pi N10, NXP Coral and checked their boot journals.

| Message | ser7 (x86) | rock pi n10 | coral | Status |
|---|---|---|---|---|
| `Unknown group 'clock'` | 0 | 0 | 0 | **fixed** |
| alsa `NAME="snd/%k"` | 0 | 0 | 0 | **fixed** |
| tmpfiles `/root` | 0 | 0 | 0 | **fixed** |
| PulseAudio cookie (4 lines) | 4 | 4 | 4 | **still on every boot** |
| bluez5 `GetManagedObjects()` | 0 | 0 | 0 | **not reproduced** (Bluetooth is running on all three) |
| xkb `XKB-679` | 3 | 3 | 3 | PR #11883 fixes it |
| Kodi D-Bus `ServiceUnknown` | ConsoleKit, UPower ×2, UDisks2, UDisks | info line only | info line only | **still open**; debug-level lines don't show |